WeActStudio/WeActStudio.EpaperModule #1: Can you please provide a wiring diagram for using on an Arduino or ESP32

Corruptsector, 2022-12-14:
I have tried to connect, but the pins are labeled for i2c, but a lot of the other example are SPI. 

Can i  use SPI to interface with this board?

WeActStudio, 2022-12-15:
Yes, it only support spi

fourgood, 2022-12-17:
But WHY are the pins labeled for i2c, if the display only supports SPI ???

tursoe, 2022-12-19:
Yes, a chart would be nice.
It's easy to find out yourself, but faster if a diagram was provided for both ESP32 and Arduino Uno.

fourgood, 2022-12-19:
I need it for ESP8266 (sry for hijacking this thread)

Corruptsector, 2022-12-27:
> Yes, a wiring diagram would be nice. It's easy to find out yourself, but faster if a diagram was provided for both ESP32 and Arduino Uno.

Did you get it working?

Im trying to get working with ESPHOME. I tried using SDA as MOSI and SCL as CLK, but no joy. 

`spi:
  clk_pin: GPIO016
  mosi_pin: GPIO017

display:
  - platform: waveshare_epaper
    cs_pin: GPIO018
    dc_pin: GPIO021
    busy_pin: GPIO034
    reset_pin: GPIO033
    model: 2.13in-ttgo-b73
    full_update_every: 2
    lambda: |-
      it.print(0, 0, id(roboto), "Hello World!");`

Corruptsector, 2022-12-27:
I think:
SDA = data = MOSI
SCL = clock = SCK

alinelena, 2022-12-31:
> Did you get it working?
>
> Im trying to get working with ESPHOME. I tried using SDA as MOSI and SCL as CLK, but no joy.
>
> ```
> spi:
>   clk_pin: GPIO016
>   mosi_pin: GPIO017
>
> display:
>   - platform: [...]

I doubt this will work... while the the display I think at base is the same (gooddisplay) things are different... did you get the display working with the esp32 example? I have a similar issue see #2

alinelena, 2022-12-31:
> Did you get it working?
>
> Im trying to get working with ESPHOME. I tried using SDA as MOSI and SCL as CLK, but no joy.
>
> ```
> spi:
>   clk_pin: GPIO016
>   mosi_pin: GPIO017
>
> display:
>   - platform: [...]

also I suggest to try as model 2.13in-ttgo-dke this seems to match the number that weact is using.

you can check the example folder and the lib for the exact numbers https://github.com/ZinggJM/GxEPD2

Corruptsector, 2023-01-04:
> I doubt this will work... while the the display I think at base is the same (gooddisplay) things are different... did you get the display working with the esp32 example? I have a similar issue see #2

No Sorry i didnt. It showed my hello world, but then started flashing and showing heaps of noise. So i guess my pinout is correct, but this model didnt work in ESPHOME

Corruptsector, 2023-01-04:
> also I suggest to try as model 2.13in-ttgo-dke this seems to match the number that weact is using.
>
> you can check the example folder and the lib for the exact numbers https://github.com/ZinggJM/GxEPD2

Im tried all the models in ESPhome without any luck. But ill try again.

alinelena, 2023-01-04:
> No Sorry i didnt. It showed my hello world, but then started flashing and showing heaps of noise. So i guess my pinout is correct, but this model didnt work in ESPHOME

indeed that indicates the screen is working. a link to the example you tried and the pinout you used will be useful. also to double check you are using bw only model.
i am in the same boat as you but mine does not even turn out

alinelena, 2023-01-04:
the pinout from the esp32 example is 

|epd | esp32|
VCC   VCC
GND  GND
SDA(MOSI)  23
SCL(SCK) 18
CS             5
D/C           0
RES           2
BUSY        15

alinelena, 2023-01-14:
just to update with the above connection and following esphome i got the 2.13 inch bw working

```
substitutions:
  devicename: boron
  update: 900s

esphome:
  name: $devicename
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${devicename} Fallback Hotspot
    password: "MLO4IA0MZb10"

captive_portal:

# Enable logging
#logger:

# Enable Home Assistant API
#api:
#  password: !secret ota_password

ota:
  password: !secret ota_password

text_sensor:
  - platform: wifi_info
    ip_address:
      id: my_ip
      name: "ip"
      internal: true

sensor:
  - platform: wifi_signal
    name: "wifi signal"
    id: my_rssi
    internal: true
    update_interval: ${update}

spi:
  id: bus_a
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

font:
  - file: "droid.ttf"
    id: my_font
    size: 10

display:
  - platform: waveshare_epaper
    spi_id: bus_a
    model: "2.13in-ttgo-dke"
    reset_pin: GPIO02
    cs_pin: GPIO05
    dc_pin: GPIO00
    busy_pin: GPIO15
    rotation: 90
    full_update_every: 60
    update_interval: 120s
    lambda: |-
      it.printf(0, 10, id(my_font), "ip: %s",id(my_ip).state.c_str());
      it.printf(0, 25, id(my_font), "rssi: %.2f",id(my_rssi).state);
      it.printf(0, 40, id(my_font), "Hello world!");

```

deepak-gunasekaran, 2023-01-26:
Does it work with raspberry pi? I could not make it work with waveshare 2.q3 display python code. Can you please help?

MarAlMe, 2023-04-07:
I made an wiring diagram for an esp8266, in my case a wemos di mini v4. The VBUS is 5V.

[image]

MarAlMe, 2023-04-09:
I got the display working with my ESP32-S2 (wemis d1 mini) and made a wiring diagram for it too. The VBUS is 5V.

[image]

MarAlMe, 2023-04-09:
I made an wiring diagram for an esp8266 (wemos d1 mini v4) and an esp32-s2 (wemos d1 mini). The VBUS is 5V.

// ESP8266 CS=15,SCL(SCK)=14,SDA(MOSI)=13,BUSY=16,RST=5,DC=4

[image]

// ESP32-S2 CS=34,SCL(SCK)=36,SDA(MOSI)=35,BUSY=3,RST=9,DC=38

[image]

OperatorB, 2023-07-22:
Thanks for the wiring diagram! By the way what tool do you @MarAlMe use to make this pretty diagram above, it looks amazing?

MarAlMe, 2023-07-31:
Thank you @OperatorB, its just Gimp 😎

Merikei, 2025-05-20:
Was able to get the 2.9″ B&W model to work on ESPHome with `model: "2.90inv2-r2"` (my first attempt, `"2.90in-dke"` did NOT work).

Though I'm not able to get Black,White&Red to work properly

Maxklos, 2025-07-05:
2.9 Inch B/W/R With an M5 Stamp C3 & ESPHome:

2.90in
2.90in-dke
2.90inv2
2.90inv2-r2
2.90in-b
2.90in-bV3

| Board| Status| 
| -------------- | -------------- | 
|  |  | 
|  |  |
|  |  |
|  |  |
|  |  |