WeActStudio/WeActStudio.EpaperModule #10: How to use it in a python project ?

Siryak, 2023-09-01:
I have some knowledge in C, way less in Python, but I want to use this display on a Raspberry Pico, using Python.
Do you have some libs for Python? Or can I have any help to do it?

playlogo, 2023-09-08:
You can use the [Adafruit SSD1680 driver](https://docs.circuitpython.org/projects/ssd1680/en/latest/). It only works on adafruit's circuit python, but is very well documented.

loonglade, 2023-09-12:
> You can use the [Adafruit SSD1680 driver](https://docs.circuitpython.org/projects/ssd1680/en/latest/). It only works on adafruit's circuit python, but is very well documented.

Is this for the 2.13 or the 2.9 or both? I have the 2.13 and I'm trying to use it on the raspberry pi zero w, but I can't figure it out even with that adafruit driver.

loonglade, 2023-09-13:
@Siryak I was able to get it to work with CircuitPython's Blinka using the EPD, SSD1680 and DisplayIO libraries and following those guides: 

https://learn.adafruit.com/2-13-in-e-ink-bonnet/usage
https://docs.circuitpython.org/projects/ssd1680/en/latest/index.html

Siryak, 2023-09-15:
> @Siryak I was able to get it to work with CircuitPython's Blinka using the EPD, SSD1680 and DisplayIO drivers and following those guides:
> 
> https://learn.adafruit.com/2-13-in-e-ink-bonnet/usage https://docs.circuitpython.org/projects/ssd1680/en/latest/index.html

@loonglade 
Firstly, thanks a lot for these links.

I have followed the instruction of the first link you've sent, and get some “artifacts”.
[image]

Here's my pinout :
Busy = 11
Res = 13
D/C = 15
CS  = 24
SCL = 23
SDA = 19
GND = 20
VCC = 17

And my code is exactly the same as in the [link](https://learn.adafruit.com/2-13-in-e-ink-bonnet/usage) :
```python
import digitalio
import busio
import board
from adafruit_epd.epd import Adafruit_EPD

spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
ecs = digitalio.DigitalInOut(board.CE0)
dc = digitalio.DigitalInOut(board.D22)
rst = digitalio.DigitalInOut(board.D27)
busy = digitalio.DigitalInOut(board.D17)
srcs = None

from adafruit_epd.ssd1680 import Adafruit_SSD1680
display = Adafruit_SSD1680(122, 250, spi, cs_pin=ecs, dc_pin=dc, sramcs_pin=srcs,
                          rst_pin=rst, busy_pin=busy)

display.fill(Adafruit_EPD.WHITE)
 
display.fill_rect(0, 0, 50, 60, Adafruit_EPD.BLACK)
display.hline(80, 30, 60, Adafruit_EPD.BLACK)
display.vline(80, 30, 60, Adafruit_EPD.BLACK)
 
display.display()
```

Do you know from where it can come from ?

Siryak, 2023-09-15:
Just use the SSD1675 driver from Adafruit, works for me.

closed 2023-09-15 by Siryak

jvlobo, 2024-10-19:
> > @Siryak I was able to get it to work with CircuitPython's Blinka using the EPD, SSD1680 and DisplayIO drivers and following those guides:
> > https://learn.adafruit.com/2-13-in-e-ink-bonnet/usage https://docs.circuitpython.org/projects/ssd1680/en/latest/index.html
> 
> @loonglade Firstly, thanks a lot for these links.
> 
> I have followed the instruction of the first link you've sent, and get some “artifacts”. ![1694770226437](https://private-user-images.githubusercontent.com/85520924/268240180-134ac999-4830-4366-8b20-f2171606920b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkzNDIyMjQsIm5iZiI6MTcyOTM0MTkyNCwicGF0aCI6Ii84NTUyMDkyNC8yNjgyNDAxODAtMTM0YWM5OTktNDgzMC00MzY2LThiMjAtZjIxNzE2MDY5MjBiLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE5VDEyNDUyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA1OTE2ZWIxMTA1MmY0N2QzODYzMjEzZGUxN2M3NTMxZTJjODA1MThkNjBiNTFjYjkyNDMzZDM2ZDNlZWY1MDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2t-Jg-8UA19WCV_zCPkw2ZeA6Lm3XNhY7U3rlYBp1a8)
> 
> Here's my pinout : Busy = 11 Res = 13 D/C = 15 CS = 24 SCL = 23 SDA = 19 GND = 20 VCC = 17
> 
> And my code is exactly the same as in the [link](https://learn.adafruit.com/2-13-in-e-ink-bonnet/usage) :
> 
> ```python
> import digitalio
> import busio
> import board
> from adafruit_epd.epd import Adafruit_EPD
> 
> spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
> ecs = digitalio.DigitalInOut(board.CE0)
> dc = digitalio.DigitalInOut(board.D22)
> rst = digitalio.DigitalInOut(board.D27)
> busy = digitalio.DigitalInOut(board.D17)
> srcs = None
> 
> from adafruit_epd.ssd1680 import Adafruit_SSD1680
> display = Adafruit_SSD1680(122, 250, spi, cs_pin=ecs, dc_pin=dc, sramcs_pin=srcs,
>                           rst_pin=rst, busy_pin=busy)
> 
> display.fill(Adafruit_EPD.WHITE)
>  
> display.fill_rect(0, 0, 50, 60, Adafruit_EPD.BLACK)
> display.hline(80, 30, 60, Adafruit_EPD.BLACK)
> display.vline(80, 30, 60, Adafruit_EPD.BLACK)
>  
> display.display()
> ```
> 
> Do you know from where it can come from ?
> 
> EDIT : Seems it work perfectly with the ssd1675 driver.

Hello and thank you for providing information.
I'm trying to get my WeAct epaper display working with CircuitPython on the Raspberry Pi Pico but I'm having some issues.

What I don't understand from your message is your pinout.
You say you have GND on PIN 20 and VCC on PIN 17, but those pins does not correspond to power lines of the Pico, right? Am I missing something?

Thanks!

loonglade, 2024-10-20:
That's right @jvlobo, @Siryak is using the wrong VCC and GND pins. I'm confused as why "20" for VCC as neither GPIO 20 nor PIN 20 is VCC... Same with GND.

Hopefully he didn't fry his board doing that.

jvlobo, 2024-10-27:
> @Siryak I was able to get it to work with CircuitPython's Blinka using the EPD, SSD1680 and DisplayIO drivers and following those guides:
> 
> https://learn.adafruit.com/2-13-in-e-ink-bonnet/usage https://docs.circuitpython.org/projects/ssd1680/en/latest/index.html

Do you have, by any chance, the code you wrote to get the WeActStudio display working with the Raspberry Pi Pico?

Thank you!

loonglade, 2024-10-28:
[epd_ssd1680.py.zip](https://github.com/user-attachments/files/17542912/epd_ssd1680.py.zip)
I've attached my file, but it's been a while and I don't remember what I had done, but I believe you could find all the information needed to get that code from the links I had provided. You'll also probably need to:

1. change the image name
2. make sure the pin in the code reflects the pin on the board and eink display
3. get all the dependencies import at the top of the file

Hope you get something out of this.

jvlobo, 2024-10-28:
> [epd_ssd1680.py.zip](https://github.com/user-attachments/files/17542912/epd_ssd1680.py.zip) I've attached my file, but it's been a while and I don't remember what I had done.
> 
> I believe you could find all the information needed to get that code from the links I had provided.
> 
> You'll also probably need to:
> 
> 1. change the image name
> 2. make sure the pin in the code reflects the pin on the board and eink display
> 3. get all the dependencies import at the top of the file
> 
> Hope you get something out of this.

Thanks a lot!! very helpful :)