WeActStudio/WeActStudio.EpaperModule #12: Current consumption in hybernate mode

nitescuc, 2023-12-09:
Hello, I'm using a 2.13" BW module with a ESP32, Arduino Framework and pretty much followed the example provided with GxEPD2 lib. I have the expected result except that I call `display.hibernate()` and I measure a 500uA power consumption where I would expect much less from the module. Does this number seem normal? What is the expected current in power off mode? And in power on? I don't seem to find the electrical caracteristics in the documents in this repo. Thanks

nitescuc, 2023-12-11:
So, for a low power application it could make sense to power the module from the ESP32 itself using a GPIO pin I suppose?

WeActStudio, 2023-12-12:
This is the result of our test, using the EpaperModuleTest_AT32F403A routine
1. Sleep Mode1,VCC=5V,IO=3.3V, sleep current is 6uA
2. Sleep Mode1,VCC=3.3V,IO=3.3V, sleep current is 55uA

[image]

nitescuc, 2023-12-12:
Thanks, I now get the same result of roughly 55uA using this initialization command:
`display.init(115200,true,2,true);` 
instead of   
`display.init(115200,true,50,false);`
with ESP32, Arduino Framework, GxEPD2 lib. The ePaper's module pins are connected to ESP32 without any other component to pull-up or pull-down, etc.
I guess you might want to update your sample code ?

closed 2023-12-12 by nitescuc