WeActStudio/WeActStudio.EpaperModule #14: Info about working MCUs

maxholgasson, 2024-01-15:
Hi, does a 328P based Arduino also works with the 2.9" display?

WeActStudio, 2024-01-16:
Should not work, io needs 3.3V level

---- Replied Message ----
| From | ***@***.***> |
| Date | 01/16/2024 05:14 |
| To | ***@***.***> |
| Cc | ***@***.***> |
| Subject | [WeActStudio/WeActStudio.EpaperModule] Info about working MCUs (Issue #14) |

Hi, does a 328P based Arduino also works with the 2.9" display?

maxholgasson, 2024-01-16:
> Should not work, io needs 3.3V level

That's of course no issue, a level shifter is used for this of course. I'm wondering about the library etc

WeActStudio, 2024-01-18:
Sorry, it hasn't been tested.

---- Replied Message ----
| From | ***@***.***> |
| Date | 01/16/2024 23:54 |
| To | ***@***.***> |
| Cc | WeAct ***@***.***>***@***.***> |
| Subject | Re: [WeActStudio/WeActStudio.EpaperModule] Info about working MCUs (Issue #14) |

Should not work, io needs 3.3V level

That's of course no issue, a level shifter is used for this of course. I'm wondering about the library etc