WeActStudio/WeActStudio.EpaperModule #16: Unable to connect to 2.9" 3C e-paper or run it with example code with GxEPD2_290_C90c on nodemcu esp-12f

deltamelter, 2024-03-19:
I trying to run the sample code on esp8266 (nodemcu 12f) with a WeactStudio 2.9 B/W/R epaper display.
Firstly it wouldn't compile without adding `#include "GxEPD2_display_selection_new_style.h"` Is thins wrong?  why is is not included in the example ino in the first place?
I have wired up using the same pins as I successfully used with MH ET BW e-paper and also the same as in the example.   
`GxEPD2_DISPLAY_CLASS<GxEPD2_DRIVER_CLASS, MAX_HEIGHT(GxEPD2_DRIVER_CLASS)> display(GxEPD2_DRIVER_CLASS(/*CS=D8*/ EPD_CS, /*DC=D3*/ 0, /*RST=D4*/ 2, /*BUSY=D2*/ 4));`

All I get is serial monitor:
```
 ets Jan  8 2013,rst cause:2, boot mode:(7,6)

waiting for host
```
Is the display broken?   Is there anything I can test?  Am I doing something wrong

dajer93, 2024-04-04:
Hey @deltamelter, what is the problem exactly? Did the compilation & upload to the ESP8266 complete successfully? If not, disconnect the display and try uploading again. After the upload is done, you can connect the display again.

deltamelter, 2024-04-05:
> Hey @deltamelter, what is the problem exactly? Did the compilation & upload to the ESP8266 complete successfully? If not, disconnect the display and try uploading again. After the upload is done, you can connect the display again.

I think some pin conflict with the nodemcu I was using.
tried a esp-wroom-02 device with same pinout and is worked fine with the arduino sample
esphome can't work on the wroom-02 some memory conflict when trying to initialise the display driver
esp32 works fine on both arduino sample and esphome

dajer93, 2024-04-06:
Could you please provide your code and the way you connected display to your pins? I'm not sure which example you're using to be frank.