WeActStudio/WeActStudio.EpaperModule #20: Current consumption in hybernate mode again

aerin, 2024-05-19:
Yes, I already read the old thread https://github.com/WeActStudio/WeActStudio.EpaperModule/issues/12. But I can't reduce consumption in hybernate mode less than 500uA. Replacing display.init(115200,true,2,true); with display.init(115200,true,50,false); didn't help me, current consumption is exactly 0.5mA.

P.S. Because I send the ESP32 into deep sleep along with the display, then I needed to install a 100k pull-up resistor to RST.

patcher-ms, 2025-04-02:
@aerin @ilarth sorry for the mention. Did you ever manage to achieve low deep sleep current?

To reduce the variables at play I tried using the ESP32C3 example (I am using a Xiao ESP32C3) and the [Arduino example form this repo](https://github.com/WeActStudio/WeActStudio.EpaperModule/blob/master/Example/EpaperModuleTest_Arduino_ESP32C3/EpaperModuleTest_Arduino_ESP32C3.ino). The only changes I made were changing the values of the GPIOs for my board (that's correct because I can see the sample drawn to the display) and added `esp_deep_sleep_start()` to also put the ESP32 in deep sleep.

I tried to:
* change the `pulldown_rst_mode` parameter of `display.init(...)` but neither true not false helped;
* hold the GPIO values with   `gpio_hold_en(...)` and `gpio_deep_sleep_hold_en()` before putting the ESP32 in deep sleep but again that didn't help;
* adding pull-up resistors to the `reset` pin and that still didn't help.

Despite attempting all of the above I am always seeing a current of about 650uA

patcher-ms, 2025-04-03:
Earlier I chatted with WeAct Studio and they told me to connect VCC of the EPD to a source > 3.3V. I tried it and the difference is measurable. I haven't had a chance to look into why.

The current consumption of both Xiao ESP32C3 and the 4.2" display with both on deep sleep changed from ~650uA when I connected VCC of the EPD to the 3V3 pin on the board to ~105uA when I connected the EPD VCC to the power supply.