WeActStudio/WeActStudio.EpaperModule #21: Question regarding the partial update procedure

avsaase, 2024-06-03:
Hi, I'm writing a new Rust driver for these displays and I'm having trouble figuring out the correct procedure for partial updates. For now I'm working with the 2.9 inch BW display.

My current approach, after initialization, is this:
1. Use the `0x44` and `0x45` commands to set the bounds of the window that I want to update. Use the `0x4E` and `0x4F` commands to set the pointers to the x and y positions at the start of the window.
2. Write the data using `0x24`. For example, to display a 16x16 black square I send 32 times `0xFF`.
3. Display the frame with `0x22` `0xFF` (display mode 2) and `0x20`. 

This works for the first time but if I repeat this at another position to draw a another square I get all kinds of strange duplication, ghosting of the square in different parts of the display. The only way I'm able to avoid this is by first writing all `0xFF` to the full frame buffer (after sending the correct commands to address the full buffer). But this only displays one square at a time which is not what I want.

Another problem is that on each partial update the border displays a sort of flicker. Is there a way to get rid of that? I tried setting a different border LUT using `0x3C`. Some settings made a border permanently black, other didn't make a difference.

I consulted the datasheet and the Arduino driver used in the examples but I wasn't able to solve the problem. Can you help me understand the right procedure?

WeActStudio, 2024-06-05:
Hello, Please check out our C language routines.

avsaase, 2024-06-05:
Can you give a link? I only found examples that use the GxEPD2 library. I looked at the library code and found it uses commands and values that are not documented in the datasheet so I'm not sure it's correct.

avsaase, 2024-06-08:
I managed to figure it out. You can find my code here: https://github.com/avsaase/weact-studio-epd

closed 2024-06-08 by avsaase