WeActStudio/WeActStudio.EpaperModule #23: change click and data pins on ESP32 possible?

pcace, 2024-07-17:
Hi there, i am trying to change the provided example with a 2.9'' module with an ESP32 DevKItV1. 
the unchanged example works, if i wire the 
SDA = data = MOSI --> 23
SCL = clock = SCK --> 18

But i need the ports otherwise. Is there any chance to change the ports to the proper SDA 21 and SCL 22 ports? 

i just cannot find a way to do that. 

Any help here? 

Thanks a lot!!

danzelbg, 2025-02-12:
I was able to accomplish that via initializing the SPI interface assigning desired GPIO.
Simply alter the example provided. 
```
#include <GxEPD2_BW.h>
#include <GxEPD2_3C.h>
#include <Fonts/FreeMonoBold9pt7b.h>
#include <SPI.h>

 // GDEH0154D67 200x200, SSD1681
GxEPD2_BW<GxEPD2_154_D67, GxEPD2_154_D67::HEIGHT> display(GxEPD2_154_D67(/*CS=10*/ 10, /*DC=*/13, /*RES=*/46, /*BUSY=*/9));

SPIClass spi;

void setup()
{
  // ....
  spi.end(); // release standard SPI pins
  spi.begin(12, 13, 11, 10); // map and init SPI pins SCK(12), MISO(13), MOSI(11), SS(10)

  display.epd2.selectSPI(spi, SPISettings(4000000, MSBFIRST, SPI_MODE0));
  display.init(115200, true, 50, false);
  
  // .. the rest
}

```

If your app was built with ESP-IDF, you can check if this link might help 
https://github.com/espressif/esp-bsp/tree/master/components/lcd/esp_lcd_ssd1681

danzelbg, 2025-02-12:
I was able to accomplish that by reinitializing the SPI interface assigning desired GPIOs.
Simply alter the example provided. 
```ino

#define ENABLE_GxEPD2_GFX 0

#include <GxEPD2_BW.h>
#include <GxEPD2_3C.h>
#include <Fonts/FreeMonoBold9pt7b.h>
#include <SPI.h>

// set your gpio pins
#define GPIO_SPI_CS 10
#define GPIO_SPI_MOSI 11
#define GPIO_SPI_MISO 13
#define GPIO_SPI_SCK 12
#define GPIO_BUSY 9
#define GPIO_RES 46

// ESP32 default SPI pins configuration CS(SS)=5,SCL(SCK)=18,SDA(MOSI)=23,BUSY=15,RES(RST)=2,DC=0

// 1.54'' EPD Module
GxEPD2_BW<GxEPD2_154_D67, GxEPD2_154_D67::HEIGHT> display(GxEPD2_154_D67(/*CS=*/ GPIO_SPI_CS, /*DC=*/GPIO_SPI_MISO, /*RES=*/GPIO_RES, /*BUSY=*/GPIO_BUSY)); 

SPIClass spi;

void setup()
{
  spi.end(); // release standard SPI pins
  spi.begin(GPIO_SPI_SCK, GPIO_SPI_MISO, GPIO_SPI_MOSI, GPIO_SPI_CS); // map and init SPI pins SCK, MISO, MOSI, SS

  display.epd2.selectSPI(spi, SPISettings(4000000, MSBFIRST, SPI_MODE0));
  display.init(115200, true, 50, false);

  // ...
}
```

If you are building your app with ESP-IDF, you can check if this link might help 
https://github.com/espressif/esp-bsp/tree/master/components/lcd/esp_lcd_ssd1681